robotology-legacy/WBI-Toolbox-controllers #1: To install or not to install?

jeljaik, 2015-02-03:
Before definitely removing the `/controllers` directory from the WBI-Toolbox and using this repo instead, I was wondering:
 1. Do you agree we should install these controllers:question: Currently, if you `make install` this project it will copy all the controllers and simulink models to `${CODYCO_SUPERBUILD_DIR}/install/share/WBI-Toolbox-controllers/controllers`
 2. What about adding it to Matlab's path when installing the WBI-Toolbox:question: 
cc @traversaro @DanielePucci @francesco-romano @iron76 

Please vote:
:+1: for yes, :-1: for no

francesco-romano, 2015-02-03:
1) :-1: 
2) :-1: 
Reasons:
1) For me controllers should be outside the codyco-superbuild and clone manually by the user (see what we do on the new robot-pc)
2) I think by default only the libraries should be added to the path. The controllers are run either locally, or is the user the one responsible to add them.

traversaro, 2015-02-03:
Sorry, I don't have any particular opinion on the issue given my limited experience with the WBI-Toolbox-controllers.

iron76, 2015-02-03:
I agree with @francesco-romano

jeljaik, 2015-02-03:
Alright then, I will prroceed to remove the controllers directory from
WBI-Toolbox  and highlight in the documentation that they have all been
moved to this new repo.

Thanks!

DanielePucci, 2015-02-03:
I agree with @francesco-romano also.

traversaro, 2015-02-05:
@francesco-romano what is the advantage of removing `WBI-Toolbox-controllers` from the codyco-superbuild ?

francesco-romano, 2015-02-05:
@traversaro  There is no a `build` associated with the repository. It is just a download.
And in a lot of uses (e.g. ours) it must reside outside the superbuild folder.
Anyway I've already talked to @jeljaik. We decided to integrate it (optionally) in the super build process

traversaro, 2015-02-05:
> @traversaro There is no a build associated with the repository. It is just a download.

Yes, and the superbuild simplifies this download step.

>  And in a lot of uses (e.g. ours) it must reside outside the superbuild folder. 

Can you elaborate more on this?

jeljaik, 2015-02-05:
In the end I decided to remove `controllers` from WBI-Toolbox and this repository is now optionally **just** downloaded by the superbuild. No installation of any sort is done, nor by codyco-superbuild, nor by Matlab.

Thanks for your comments.

Proceeded with commits:
- codyco-superbuild: https://github.com/robotology/codyco-superbuild/commit/74cb19001563148692bd37e67077e142a7c7fd40
- WBI-Toolbox-controllers: https://github.com/robotology-playground/WBI-Toolbox-controllers/commit/aeaf218b0f252b5c4f7d30b48b8fdf74cd2ca1ed
- WBI-Toolbox: https://github.com/robotology-playground/WBI-Toolbox/commit/6a1e8f0f257b4bad81a530031e069d3c6ac4f10d

closed 2015-02-05 by jeljaik