robotology-legacy/WBI-Toolbox-controllers #3: problem in running balancing controller

azadm, 2015-03-31:
When I try to run  the controller in `balancing_with_opt_of_tau`, MATLAB crashes and says that it needs to be closed!  
Here is what I get in the terminal:
````
All parameters have been checked and passed correctly
Yarp read block will have: 23outputs
[INFO][mdlInitializeSize] BLOCK TYPE IS: 3
[INFO][mdlInitializeSizes] wbiConfFileName yarpWholeBodyInterface.ini 
[INFO][mdlInitializeSizes] wbiConfFileFullPath is: /Users/azadm/Documents/codyco-superbuild/build/install/share/codyco/robots/icubGazeboSim/yarpWholeBodyInterface.ini
[INFO][mdlInitializeSizes] List of properties for yarpWholeBodyInterface was interpreted correctly
[INFO][mdlInitializeSizes] ROBOT_DOF: 23 
[INFO][mdlInitializeSizes] Options set

[INFO][mdlInitializeSize] BLOCK TYPE IS: 2
[INFO][mdlInitializeSizes] wbiConfFileName yarpWholeBodyInterface.ini 
[INFO][mdlInitializeSizes] wbiConfFileFullPath is: /Users/azadm/Documents/codyco-superbuild/build/install/share/codyco/robots/icubGazeboSim/yarpWholeBodyInterface.ini
[INFO][mdlInitializeSizes] List of properties for yarpWholeBodyInterface was interpreted correctly
[INFO][mdlInitializeSizes] ROBOT_DOF: 23 
[INFO][mdlInitializeSizes] Options set

[INFO][mdlInitializeSize] BLOCK TYPE IS: 0
[INFO][mdlInitializeSizes] wbiConfFileName yarpWholeBodyInterface.ini 
[INFO][mdlInitializeSizes] wbiConfFileFullPath is: /Users/azadm/Documents/codyco-superbuild/build/install/share/codyco/robots/icubGazeboSim/yarpWholeBodyInterface.ini
[INFO][mdlInitializeSizes] List of properties for yarpWholeBodyInterface was interpreted correctly
[INFO][mdlInitializeSizes] ROBOT_DOF: 23 
[INFO][mdlInitializeSizes] Options set

[INFO][mdlInitializeSize] BLOCK TYPE IS: 11
[INFO][mdlInitializeSizes] wbiConfFileName yarpWholeBodyInterface.ini 
[INFO][mdlInitializeSizes] wbiConfFileFullPath is: /Users/azadm/Documents/codyco-superbuild/build/install/share/codyco/robots/icubGazeboSim/yarpWholeBodyInterface.ini
[INFO][mdlInitializeSizes] List of properties for yarpWholeBodyInterface was interpreted correctly
[INFO][mdlInitializeSizes] ROBOT_DOF: 23 
[INFO][mdlInitializeSizes] Options set

[INFO][mdlInitializeSize] BLOCK TYPE IS: 0
[INFO][mdlInitializeSizes] wbiConfFileName yarpWholeBodyInterface.ini 
[INFO][mdlInitializeSizes] wbiConfFileFullPath is: /Users/azadm/Documents/codyco-superbuild/build/install/share/codyco/robots/icubGazeboSim/yarpWholeBodyInterface.ini
[INFO][mdlInitializeSizes] List of properties for yarpWholeBodyInterface was interpreted correctly
[INFO][mdlInitializeSizes] ROBOT_DOF: 23 
[INFO][mdlInitializeSizes] Options set

[INFO][mdlInitializeSize] BLOCK TYPE IS: 13
[INFO][mdlInitializeSizes] wbiConfFileName yarpWholeBodyInterface.ini 
[INFO][mdlInitializeSizes] wbiConfFileFullPath is: /Users/azadm/Documents/codyco-superbuild/build/install/share/codyco/robots/icubGazeboSim/yarpWholeBodyInterface.ini
[INFO][mdlInitializeSizes] List of properties for yarpWholeBodyInterface was interpreted correctly
[INFO][mdlInitializeSizes] ROBOT_DOF: 23 
[INFO][mdlInitializeSizes] Options set

[INFO][mdlInitializeSize] BLOCK TYPE IS: 14
[INFO][mdlInitializeSizes] wbiConfFileName yarpWholeBodyInterface.ini 
[INFO][mdlInitializeSizes] wbiConfFileFullPath is: /Users/azadm/Documents/codyco-superbuild/build/install/share/codyco/robots/icubGazeboSim/yarpWholeBodyInterface.ini
[INFO][mdlInitializeSizes] List of properties for yarpWholeBodyInterface was interpreted correctly
[INFO][mdlInitializeSizes] ROBOT_DOF: 23 
[INFO][mdlInitializeSizes] Options set

[INFO][mdlInitializeSize] BLOCK TYPE IS: 7
[INFO][mdlInitializeSizes] wbiConfFileName yarpWholeBodyInterface.ini 
[INFO][mdlInitializeSizes] wbiConfFileFullPath is: /Users/azadm/Documents/codyco-superbuild/build/install/share/codyco/robots/icubGazeboSim/yarpWholeBodyInterface.ini
[INFO][mdlInitializeSizes] List of properties for yarpWholeBodyInterface was interpreted correctly
[INFO][mdlInitializeSizes] ROBOT_DOF: 23 
[INFO][mdlInitializeSizes] Options set

[INFO][mdlInitializeSize] BLOCK TYPE IS: 7
[INFO][mdlInitializeSizes] wbiConfFileName yarpWholeBodyInterface.ini 
[INFO][mdlInitializeSizes] wbiConfFileFullPath is: /Users/azadm/Documents/codyco-superbuild/build/install/share/codyco/robots/icubGazeboSim/yarpWholeBodyInterface.ini
[INFO][mdlInitializeSizes] List of properties for yarpWholeBodyInterface was interpreted correctly
[INFO][mdlInitializeSizes] ROBOT_DOF: 23 
[INFO][mdlInitializeSizes] Options set

[INFO][mdlInitializeSize] BLOCK TYPE IS: 8
[INFO][mdlInitializeSizes] wbiConfFileName yarpWholeBodyInterface.ini 
[... 295 lines cut ...]
yarp: Removing input from /icubGazeboSim/left_leg/stateExt:o to /model/left_leg/stateExt:i
yarp: Removing output from /model/right_leg/rpc:o to /icubGazeboSim/right_leg/rpc:i
yarp: Removing output from /model/right_leg/command:o to /icubGazeboSim/right_leg/command:i
yarp: Removing input from /icubGazeboSim/right_leg/state:o to /model/right_leg/state:i
yarp: Removing input from /icubGazeboSim/right_leg/stateExt:o to /model/right_leg/stateExt:i
[INFO][mdlStart] Succesfully exited robotConfig.
[INFO][robotStatus::robotInit] Finished robotInit
[INFO][mdlStart] Succesfully exited robotInit.
[INFO][mdlStart] FINISHED

[INFO][mdlStart] I AM GETTING UPDATED!!! 
YARP NETWORK INITIALIZED
YARP is running!!

yarp: Port /tmp/port/2 active at tcp://147.188.200.113:10101
[From] Port name will be: /y2coordinator/qdes:o
[To] Port name will be:   /tmp/port/2
Blocking? : 0
Timestamp? : 0
Autoconnect? : 1
Opening log file:  /Users/azadm/java.log.22968
Killed: 9
````

Is there any extra setting that needs to be done before running this file?

DanielePucci, 2015-03-31:
Hi,
we are in the middle of some testing. Give me 15 minutes.

DanielePucci, 2015-03-31:
Hi,
update everything (yarp, icub, codyco-superbuild, WBI-Controllers) and it should work. It works on my laptop (standing and left-and-right).

azadm, 2015-03-31:
Hi,

I updated all. Now I get an assertion error!
~~~
Assertion detected in 'controller/Subsystem/emergency stop: joint limits/Assertion' at time 0
~~~

By the way, the `gains.m` in the `icubGazeboSim` folder does not have the required parameters `qTildeMax`, `satIntegral` and `integralGains`.

DanielePucci, 2015-03-31:
Deeply sorry @azadm,
I did not push. Try now!

azadm, 2015-03-31:
Thank you @DanielePucci 
Now it works but it doesn't stabilize the robot. Is there anything else that I need to set to make it work properly?

DanielePucci, 2015-03-31:
Mmmm, that's weird. There are two possible points:

1) These versions of the controllers are not synced with Gazebo. Then, the Real Time Factor may play a role. Can you tell me the realTimeFacto on your laptop?

2) Friction. We still have numerical integration issues related with Gazebo. Then,  open icub-gazebo/icub/icub.sdf and set the tag damping (under the tree of tags dynamics axis, joint), equal to one for all joints, e.g.

        <joint name='l_hip_roll' type='revolute'>
            <child>l_hip_2</child>
            <parent>l_hip_1</parent>
            <axis>
                <xyz>-1 5.30716e-06 1.26536e-05</xyz>
                <limit>
                    <lower>-0.3491</lower>
                    <upper>1.3090</upper>
                    <effort>10000</effort>
                    <velocity>100</velocity>
                </limit>
                <dynamics>
                    <damping>1.0</damping>
                </dynamics>

azadm, 2015-03-31:
1) I am running this on my desktop so the real time factor is always above 0.9.
2) I will do that and let you know ...
3) I always start from the default initial configuration.

DanielePucci, 2015-03-31:
Ok, try to update now the repository icub-gazebo instead of changing the tags

DanielePucci, 2015-03-31:
Which version of gazebo do you have?

azadm, 2015-03-31:
Great..Now it is working perfectly. Thanks a lot @DanielePucci 

I am using Gazebo 4 by the way.

So it was all about friction in the joints?

DanielePucci, 2015-03-31:
Yep, but these issues will be addressed after the CoDYCo demo :) I suggest you to install gazebo5

azadm, 2015-03-31:
OK and good luck with the demo ;)

closed 2015-03-31 by DanielePucci