robotology-legacy/WBI-Toolbox-controllers #4: [balancing_with_opt_of_tau] Instabilities when using velocities rpovided by Gazebo

DanielePucci, 2015-04-04:
When using the option

~~~
readSpeedAccFromControlBoard
~~~

in the configuration file yarpWholeBodyInterface.ini, which enables velocity readings from Gazebo and skip the estimation by adaptive filters, the controller is not stable.

DanielePucci, 2015-04-04:
As for a unit test, I'm using the controller positionControl, which gives sinusoidal reference joint trajectories to all 23 DOFs. The robot is fixed, and I run the simulations twice, without and with the flag

~~~
readSpeedAccFromControlBoard
~~~

in the configuration file yarpWholeBodyInterface.ini. 

The velocities estimated by the adaptive filters are: 

[image]

while the velocities provided by Gazebo are

[image]

Any clue on these sharp variations? @traversaro @francesco-romano

traversaro, 2015-04-04:
Mhh I guess it could be related to ODE errors, perhaps fine tuning some parameters could mitigate this problem..

gabrielenava, 2019-02-28:
Outdated issue. Closing.

closed 2019-02-28 by gabrielenava