robotology-legacy/WBI-Toolbox-controllers #6: [WBI-Toolbox-controllers] Migration of all balancing controllers into one called "torqueBalancing"

DanielePucci, 2015-04-20:
I've attempted at unifying all balancing controllers developed so far into only one simulink model. This new controller is under the folder "torqueBalacing". Be aware that this new controller depends on the solver QPOASES, which must be installed accordingly, e.g. in a terminal type

~~~
cd $CODYCO_SUPERBUILD_ROOT
git pull
cd build
cmake .
make update-all
ccmake .
~~~

Then set  CODYCO_BUILD_QPOASES = ON, press "c" until "g" appears, then press "g", quit, and type

~~~
make
~~~

 Tell me if you have troubles with this new version of the controller in simulations @rlober @azadm @francesco-romano @jeljaik @nunoguedelha @naveenoid

traversaro, 2015-04-22:
I guess also the `QPOASES_BUILD_BINDINGS_MATLAB` option in the qpOASES cmake should be selected to be true. 

Anyway, I have just enabled qpOASES by default in the superbuild. If anyone has problems please report them. thanks.

DanielePucci, 2015-10-25:
ready to re-open the issue in the case users arise some problems.

closed 2015-10-25 by DanielePucci