robotology-legacy/WBI-Toolbox-controllers #8: torqueBalancing on right foot

hu-yue, 2015-10-23:
Hi,
We usually perform the torque balancing on single leg with the left leg.
But I wanted to try the torqueBalancing with the robot standing on the right foot.
I did the same procedures I usually do with the left leg (launching also wholeBodyDynamicsTree with --assume_fixed r_foot_dh_frame and calibrate with calibStandingRightFoot), but this does not work.
When I launch the Simulink the robot fell, i.e. the left leg fell.

Should I adjust something else?
Thanks!
@DanielePucci

jeljaik, 2015-10-23:
While @DanielePucci finds a moment to give you a more detailed answer, let me just remind you that it's not enough to specify which link is fixed in `wholeBodyDynamicsTree` but also for the `WBI-Toolbox`, since it assumes one link fixed in the robot as the inertial frame, and that link by default is the right foot. So when you switch to the other foot all the computations are affected and that's why things won't work with the changes you made.

hu-yue, 2015-10-23:
@jeljaik Yeah you're right, I forgot about that, will try again by adjusting that too!
Thanks!

DanielePucci, 2015-10-23:
Hi @hu-yue, do not forget to change the gains associated with the right/left leg. In particular, in 

https://github.com/robotology-playground/WBI-Toolbox-controllers/blob/master/controllers/torqueBalancing/robots/iCubHeidelberg01/gains.m#L80

You must switch the values in `impLeftLeg` with those in `impRightLeg`, and it shall work. Let us know, because you are probably the first one to do this test :)

@francesco-romano

DanielePucci, 2015-10-23:
@hu-yue, if it works, we can close it :)

hu-yue, 2015-10-23:
I tried as @jeljaik suggested, and changed the world ref both in the yarpWholeBodyInterface.ini and wholeBodyInterfaceToolbox.ini and it sort of works (not very stable but does not fall as before), I will try to change also the impedances and will let you know.

hu-yue, 2015-10-23:
@DanielePucci It works, but due to https://github.com/robotology/icub-support/issues/90#issuecomment-150538226, I'm not sure how good it works, as it goes on hw fault if I press the leg for a few times.

DanielePucci, 2015-10-23:
I understand. If you do not press the leg, and I assume that you have switched the gains, are the performances similar to those of the left-foot balancing?

hu-yue, 2015-10-23:
@DanielePucci Yes, I tried also the yoga, it also works as with the left foot.

DanielePucci, 2015-10-23:
Great @hu-yue, closing this issue then.

closed 2015-10-23 by DanielePucci