robotology-legacy/WBI-Toolbox-controllers #9: Problems with Clock when following the torqueBalancing instructions

iron76, 2015-10-30:
I am getting the following error
```
Error reported by S-function 'ySynchronizer' in 'torqueBalancing/Subsystem/synchronizer/synchronize':
Problems connecting with /clock/rpc. Has gazebo been launched with the clock plugin?
```
which can be solved by removing or commenting out the blocks for the synchronisation. However, none of the models in repository https://github.com/robotology-playground/icub-gazebo seems to use the clock plugin.

DanielePucci, 2015-10-30:
Hi @iron76, the torqueBalancing model uses the `ySyncronizer` block. A possibility is to launch Gazebo as specified, e.g., in

https://github.com/robotology-playground/WBI-Toolbox-controllers/blob/master/controllers/torqueBalancing/initTorqueBalancing.m#L32

i.e. 

~~~
gazebo -slibgazebo_yarp_clock.so
~~~

Tell me if it works.

DanielePucci, 2015-10-30:
P.S. I am right now on OsX, and tests on `torqueBalancing` performed on my laptop passed just yesterday. For some reasons, it seems to me that the synchronisation between the controller and the Gazebo becomes fundamental for making the controller work on OsX. @traversaro @francesco-romano @gabrielenava

iron76, 2015-11-01:
Please consider adding the `gazebo -slibgazebo_yarp_clock.so` to the instructions (https://github.com/robotology-playground/WBI-Toolbox-controllers/tree/master/controllers/torqueBalancing). In my opinion it is not that obvious that it is necessary to edit the `initTorqueBalancing.m` file before launching the controller.

DanielePucci, 2015-11-01:
Done in https://github.com/robotology-playground/WBI-Toolbox-controllers/commit/4bba0885a6079dc69fb52e96dd08e064a93843a1

closed 2015-11-01 by DanielePucci