robotology-legacy/WBI-Toolbox-controllers #10: [torqueBalancing] "Conflicts" between reference generator and torqueBalancing

hu-yue, 2015-12-03:
Hi,
I'm not sure if this issue should go here and if this is a bug or what else, however today we encountered the following problem: we launched the ```torqueBalancing``` from the matlab module on ```pc1```, and it was running without problem, then on ```pc2``` we launched another matlab session to run the ```reference generator```, and we had errors.
We stopped the ```torqueBlancing```, and tried to run only the ```reference generator```, and it run without problems.
Then I tried also to run the ```torqueBalancing``` from the c++ module, and then the ```reference generator``` from matlab, and again we got errors. But if we launch ```reference generator``` first, then no errors occur.
With the c++ module the errors appear if the ```torqueBalancing``` has been started. If it's just launched without starting from rpc, then the error also do not occur.

So it seemed to me that there were some conflicts, but I don't know how to figure out exactly where the problem is.

@kevinastein

traversaro, 2015-12-03:
You saved/copy pasted any of the errors? They can be helpful to start the debug.

hu-yue, 2015-12-04:
These are the errors printed in Matlab:
```
[INFO]created device <remote_controlboard>. See C++ class yarp::dev::RemoteControlBoard for documentation.
[INFO]RemoteControlBoard is ENABLING the writeStrict option for all commands
[INFO]created device <remote_controlboard>. See C++ class yarp::dev::RemoteControlBoard for documentation.
[ERROR]yarpWholeBodySensors::readPwms failed for timeout
[ERROR]yarpWholeBodyStates::init : estimator thread initialization failed 
[ERROR][robotStatus::robotConfig] Initializing Whole Body Interface!
[ERROR][mdlStart] in robotConfig
[INFO][mdlTerminate] Inside robot object 0x7f91f5365ad0 
[INFO][mdlTerminate] ctrl mode set
[INFO][robotStatus::~robotStatus] wbInterface in destructor: 0x7f91f4643bf0 
[INFO][robotStatus::~robotStatus] wbInterface has been closed and deleted correctly. -1 to go 
[INFO][mdlTerminate] robot deleted
[INFO][mdlTerminate] yarpWbiOptions deleted
[INFO][mdlTerminate] all variables set to NULL
[INFO][mdlTerminate] Left mdlTerminate
```

This is a screenshot of the error given by simulink:
[image]

francesco-romano, 2015-12-04:
It's quite scary that this happens so systematically...and both in matlab and c++.
The error means that the wbi couldn't read the sensors.. and interestingly the PWM which we are not using now (am I right @traversaro ?)

traversaro, 2015-12-04:
I think the `yarpWholeBodyState` systematically reads the PWM.

naveenoid, 2015-12-21:
From my understanding of yarpWholeBodySensors, it seems to be triggered around this line : 
https://github.com/robotology/yarp-wholebodyinterface/blob/master/src/yarpWholeBodySensors.cpp#L720

This implies its an issue with the IOpenLoopControl interface (i.e. the getOutputs method)

gabrielenava, 2019-02-28:
Outdated issue. Closing.

closed 2019-02-28 by gabrielenava