robotology-legacy/WBI-Toolbox-controllers #13: torqueBalancing controller won't run on Simulink 2012b (8.0)

nunoguedelha, 2016-01-18:
Since I've updated the controllers through `codyco-superbuild`, loading the torqueBalancing controller will make Matlab crash systematically. Actually, I'm running Matlab 2012b with the online licenced Simulink Toolbox 2012b. I've just noticed the controller has been saved in Simulink 8.6 (2015b), probably not compatible with Matlab2012b.
@DanielePucci could you please check this compatibility issue, and eventually the possibility to save models in a "compatibility mode"? Thanks

DanielePucci, 2016-01-19:
HI @nunoguedelha, we are about to merge a new brach into master, where there are also the versions compatible with MATLAB 2012b. May you wait a little or you need it ASAP?

nunoguedelha, 2016-01-19:
@DanielePucci sure I can wait.

DanielePucci, 2016-01-20:
Hi @nunoguedelha @misaki43, I have added in `torqueBalancing` a file that should be compatible with MATLAB 2012a. May you check if it works?

Thanks
@traversaro

misaki43, 2016-01-21:
Hi @DanielePucci , the problem is solved, Thanks!
Now I have another error but I haven't look at it yet:
>Error evaluating parameter 'UpperLimit' in
'torqueBalancing2012a/Subsystem/nonlinearImpedances/Saturation2'
Caused by:
    Undefined function 'impedencesSat' for
    input arguments of type 'double'.

I have a request even if it may be too long for you: is it possible to do video tutorial about "how to use Simulink with iCub" with some example? (I could do it after too, if I understand how it works)

DanielePucci, 2016-01-21:
So, the problem is that MATLAB 2012a is not compatible with some features. Can you tell us which MATLAB version do you have?

DanielePucci, 2016-01-21:
Tested on MATLAB 2012b.

misaki43, 2016-01-21:
My version is "MATLAB R2013a"

nunoguedelha, 2016-01-21:
I can load the model but building is falling, but it might be a problem with my libraries. I get this error:
```
Error using construct_error (line 108)
Library failed to build. Cannot continue build process.
```
and it seams to be related to `torqueBalancing2012b/Subsystem/DesValues/Read
desired_x_dx_ddx_CoM as 9 element vector` sub-module. I'm looking a bit farther...

DanielePucci, 2016-01-22:
Guys, can you update me on this?

misaki43, 2016-01-22:
Hi! 
Thanks it is better.
During the initialization I have some warnings but it is not blocked:
>Warning: Matching "From" for "Goto"
'torqueBalancing2012b/Subsystem/controller_QP/fDesNoQP'
not found 
Warning: Matching "From" for "Goto"
'torqueBalancing2012b/Subsystem/controller_QP/fDesQP'
not found 
Warning: Output port 14 of
'torqueBalancing2012b/Subsystem/DesValues/CoM
/COMFwdKin /S-Function' is not connected. 
Warning: Output port 14 of
'torqueBalancing2012b/Subsystem/DesValues/q(0)/Joint
Angles/S-Function' is not connected. 
Warning: Output port 14 of
'torqueBalancing2012b/Subsystem/Joint
Angles/S-Function' is not connected. 
Warning: Output port 14 of
'torqueBalancing2012b/Subsystem/Joint
Limits1/S-Function' is not connected. 
Warning: Output port 14 of
'torqueBalancing2012b/Subsystem/M(q) h(q,dq)
g(q)/Cent Momentum/S-Function' is not
connected. 
Warning: Output port 14 of
'torqueBalancing2012b/Subsystem/M(q) h(q,dq)
g(q)/computeh/S-Function' is not connected. 
Warning: Output port 14 of
'torqueBalancing2012b/Subsystem/M(q) h(q,dq)
g(q)/massMatrix/massMatrix/S-Function' is not
connected. 
Warning: Output port 14 of
'torqueBalancing2012b/Subsystem/Set
Torques1/S-Function' is not connected. 
Warning: Output port 14 of
'torqueBalancing2012b/Subsystem/base +
dq'/Joint Velocities/S-Function' is not
connected. 
Warning: Output port 14 of
'torqueBalancing2012b/Subsystem/base + dq'/Left
Foot Jacobian/Left Foot  Jacobian/S-Function'
is not connected. 
Warning: Output port 14 of
'torqueBalancing2012b/Subsystem/base +
dq'/Right Foot Jacobian/Right Foot
Jacobian/S-Function' is not connected. 
Warning: Output port 14 of
'torqueBalancing2012b/Subsystem/controller_QP/COMFwdKin
/S-Function' is not connected. 
Warning: Output port 14 of
'torqueBalancing2012b/Subsystem/controller_QP/Center
of Mass Jacobian/COM Jacobian/S-Function' is
not connected. 
Warning: Output port 14 of
'torqueBalancing2012b/Subsystem/controller_QP/Left
Foot Jacobian/Left Foot  Jacobian/S-Function'
is not connected. 
Warning: Output port 14 of
'torqueBalancing2012b/Subsystem/controller_QP/LeftFootFwdKin/S-Function'
is not connected. 
Warning: Output port 14 of
'torqueBalancing2012b/Subsystem/controller_QP/Right
Foot Jacobian/Right Foot  Jacobian/S-Function'
is not connected. 
Warning: Output port 14 of
'torqueBalancing2012b/Subsystem/controller_QP/RightFootFwdKin/S-Function'
is not connected. 
Warning: Output port 14 of
'torqueBalancing2012b/Subsystem/controller_QP/dJ*dq
LeftFoot/S-Function' is not connected. 
Warning: Output port 14 of
'torqueBalancing2012b/Subsystem/controller_QP/dJ*dq
RightFoot/S-Function' is not connected. 
Warning: Source
'torqueBalancing2012b/Subsystem/DesValues/
LEFT_RIGHT_FOOT_IN_CONTACT' specifies that its
sample time (-1) is back-inherited. You should
explicitly specify the sample time of sources.
You can disable this diagnostic by setting the
'Source block specifies -1 sample time'
diagnostic to 'none' in the Sample Time group
on the Diagnostics pane of the Configuration
Parameters dialog box. 
Warning: Source
'torqueBalancing2012b/Subsystem/DesValues/Read
qDes_impednces' specifies that its sample time
(-1) is back-inherited. You should explicitly
specify the sample time of sources. You can
disable this diagnostic by setting the 'Source
block specifies -1 sample time' diagnostic to
'none' in the Sample Time group on the
Diagnostics pane of the Configuration
Parameters dialog box. 
Warning: Source
'torqueBalancing2012b/Subsystem/DesValues/Read
desired_x_dx_ddx_CoM as 9 element vector '
specifies that its sample time (-1) is
back-inherited. You should explicitly specify
the sample time of sources. You can disable
this diagnostic by setting the 'Source block
specifies -1 sample time' diagnostic to 'none'
in the Sample Time group on the Diagnostics
pane of the Configuration Parameters dialog
box. 

So, a file is not found and things are not connected, but since I don't know how to use Simulink with iCub (I just read some simple Simulink tutorials), I bet it is me that forget something. 
After these warnings, a lot of update has been done and finished with 
>[INFO][mdlStart] I AM GETTING UPDATED!!! 
[INFO][ySynhcronizer] PAUSED THE SIMULATION DURING THE FIRST SIMULATION LOOP

Now, I have to understand how to use it, can you guide me?

nunoguedelha, 2016-01-22:
The "build" was failing, but this action was actually trying to generate C++ code from the model, while I just wanted to compile it in the first place.

So **compiling and running** is working for me. I also have warnings but I will look into that later and eventually rise a different tracking issue for that matter if necessary. Thanks!

nunoguedelha, 2016-01-22:
I suggest we close this issue.

francesco-romano, 2016-01-22:
Ok. Discussions on how to use the model can be continued in https://github.com/robotology-playground/WBI-Toolbox/issues/112.

Closing this..

closed 2016-01-22 by francesco-romano