robotology-legacy/WBI-Toolbox-controllers #14: "Yoga" demo tutorial

traversaro, 2016-02-03:
For education purpose it would be nice to have a tutorial explaining how to run the so-called "Yoga" demo in simulation. 

cc @DanielePucci @francesco-romano @fjandrad

DanielePucci, 2016-02-04:
Yes, though the `YOGA` is not stable yet. As soon as it is, we'll take care of pedagogic concerns.

traversaro, 2016-02-04:
"stable" with respect to what? :smile: 
Jokes apart, I was wrong in calling it a "tutorial for educational purpouse". My first concern is to share the knoledge of how to run the `YOGA` controller in the ICL group in the first place (as a first point.. where is the Simulink model?) to experiment a bit with it.. and perhaps so that everyone can give an hand on making it "stable".

lornat75, 2016-02-04:
Remember that yoga refers to the old year 3 demo, for some people (like me) it is misleading. What about yoga++ ?

traversaro, 2016-02-04:
I agree, I was confused by that a lot of time, that's why I called it `so-called "Yoga"`.

DanielePucci, 2016-02-04:
I mean, for any epsilon greater than zero there exists a delta of epsilon such that if the system's initial condition belongs to the hyper sphere of radius delta of epsilon, then the system's trajectories will belong to the hyper sphere of radius epsilon.

DanielePucci, 2016-02-04:
Jokes apart, here's the simulink model: 

`https://github.com/robotology-playground/WBI-Toolbox-controllers/tree/torqueBalancingJointControl/controllers/torqueBalancingWBT`

gabrielenava, 2019-02-28:
Tutorials on our demos are in [whole-body-controllers](https://github.com/robotology/whole-body-controllers). Closing.

closed 2019-02-28 by gabrielenava