robotology-legacy/WBI-Toolbox-controllers #15: Write a description of each controller in the main README

traversaro, 2016-02-04:
And also add a link to the README of each controller. 
See https://github.com/robotology-playground/WBI-Toolbox-controllers/pull/11#issuecomment-165722183 .

gabrielenava, 2016-02-05:
Done! I'll wait for suggestions before closing this issue

traversaro, 2016-02-05:
Thanks! In the future feel free to link the relevant commit in the issue, in this case:
https://github.com/robotology-playground/WBI-Toolbox-controllers/commit/0b2a8d3d124b7e5a6166320af4efcdaa1000c565

In this way if someone wants to check out the commit can just click on it and does not need to search it on the repo (especially if the commit is not on the master branch). 

Alternatively, you can refer to the issue in the commit message, either using the `#15` notation or using the full URL of the issue: https://github.com/robotology-playground/WBI-Toolbox-controllers/issues/15 . In this way a link to the commit will be automatically added to the issue.

gabrielenava, 2017-09-07:
closing because this documentation is now outdated, and opening a new one asking for a general update of the documentation in this repo.

closed 2017-09-07 by gabrielenava