robotology-legacy/WBI-Toolbox-controllers #21: [BUG] README with broken links

iron76, 2016-04-27:
The README of the project has a broken link in the torqueBalancingJc README.

DanielePucci, 2016-04-27:
@gabrielenava may you fix these links?

gabrielenava, 2016-04-28:
Done at https://github.com/robotology-playground/WBI-Toolbox-controllers/commit/b985d6da836ed37aed34ad179c16c20c64dc89ad . Actually, the current version of the joint space controller is under pull request, so the link is now pointing the torqueBalancingJointControl/README.md in the ModifJointControl branch. It will be updated again after merging the branch into master

iron76, 2016-04-28:
There is still some issue in the link torqueBalancingJc README.

gabrielenava, 2016-04-28:
this commit https://github.com/robotology-playground/WBI-Toolbox-controllers/commit/2f2597da70529e081920e7133971889653086bad on the master branch and this one https://github.com/robotology-playground/WBI-Toolbox-controllers/commit/55aa79e9f45d1cd1b374069aa13d8fdc0b18f42a  on the ModifJointControl branch should definitely fix the bug

gabrielenava, 2017-08-28:
Closing this issue

closed 2017-08-28 by gabrielenava