robotology-legacy/WBI-Toolbox-controllers #27: MATLAB setenv/getenv does not play nice with NetworkBase::getEnvironment

traversaro, 2016-06-07:
http://www.yarp.it/classyarp_1_1os_1_1NetworkBase.html#a283af07d20c62b6eb9826f9bda68c3f2

DanielePucci, 2016-06-08:
Hi @traversaro , what do you mean by "does not play nice"?

traversaro, 2016-06-08:
Enviromental variables set by matlab (as in https://github.com/robotology-playground/WBI-Toolbox-controllers/blob/d7a21f545d1ba75ba68679ab37d0c23e9225da56/controllers/wholeBodyImpedance/init_impedanceControl.m#L30) are not seen by the underlying C++ code in the `WB-Toolbox`. Still needs to be investigated.

DanielePucci, 2016-06-08:
Got it. Thanks.

francesco-romano, 2016-10-28:
It seems to happen only in Windows

gabrielenava, 2019-02-28:
`setenv` and `getenv` are no more used in [whole-body-controllers](https://github.com/robotology/whole-body-controllers). Closing.

closed 2019-02-28 by gabrielenava