robotology-legacy/WBI-Toolbox-controllers #30: Occasional Matlab crash when initializing torqueBalancing

S-Dafarra, 2016-08-22:
Here the dump
```
------------------------------------------------------------------------
          std::terminate() detected at Mon Aug 22 14:36:58 2016
------------------------------------------------------------------------

Configuration:
  Crash Decoding      : Disabled
  Crash Mode          : continue (default)
  Current Graphics Driver: Intel Open Source Technology Center Mesa DRI Intel(R) HD Graphics 5500 (Broadwell GT2)  Version 3.0 Mesa 10.5.9
  Current Visual      : 0x20 (class 4, depth 24)
  Default Encoding    : UTF-8
  GNU C Library       : 2.19 stable
  Host Name           : iiticublap042
  MATLAB Architecture : glnxa64
  MATLAB Root         : /usr/local/MATLAB/R2016a
  MATLAB Version      : 9.0.0.341360 (R2016a)
  OpenGL              : hardware
  Operating System    : Linux 3.19.0-66-generic #74~14.04.1-Ubuntu SMP Tue Jul 19 19:56:11 UTC 2016 x86_64
  Processor ID        : x86 Family 6 Model 61 Stepping 4, GenuineIntel
  Virtual Machine     : Java 1.7.0_60-b19 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode
  Window System       : The X.Org Foundation (11701000), display :0

Fault Count: 1

Abnormal termination:
std::terminate()

Register State (captured):
  RAX = 0000000000000020  RBX = 00007f72a890a7e8
  RCX = 0000000000000002  RDX = 00007f72a88fac20
  RSP = 00007f70ca7edff0  RBP = 00007f70ca7ee3b0
  RSI = 00007f72a86b13b8  RDI = 00007f70ca7ee000

   R8 = 0000000000000000   R9 = 00007f72ad309fb0
  R10 = 00007f72ac6a887e  R11 = 00007f72ac6a887f
  R12 = 00007f72a891a878  R13 = 00007f70ca7ef070
  R14 = 00007f70ca7ef050  R15 = 00007f70ca7ef430

  RIP = 00007f72a861e59e  EFL = 0000000000000002

   CS = 0001   FS = 0000   GS = 0000

Stack Trace (captured):
[  0] 0x00007f72a861e59e    /usr/local/MATLAB/R2016a/bin/glnxa64/libmwfl.so+01082782 _ZN2fl4diag5linux6x86_6412context_base12capture_dataEv+00000030
[  1] 0x00007f72a861ae0a    /usr/local/MATLAB/R2016a/bin/glnxa64/libmwfl.so+01068554
[  2] 0x00007f729d5771e5   /usr/local/MATLAB/R2016a/bin/glnxa64/libmwmcr.so+00913893
[  3] 0x00007f729d578fa0   /usr/local/MATLAB/R2016a/bin/glnxa64/libmwmcr.so+00921504
[  4] 0x00007f729d579088   /usr/local/MATLAB/R2016a/bin/glnxa64/libmwmcr.so+00921736
[  5] 0x00007f729d579453   /usr/local/MATLAB/R2016a/bin/glnxa64/libmwmcr.so+00922707
[  6] 0x00007f729d5794c3   /usr/local/MATLAB/R2016a/bin/glnxa64/libmwmcr.so+00922819
[  7] 0x00007f72ad0806d6           /usr/lib/x86_64-linux-gnu/libstdc++.so.6+00386774
[  8] 0x00007f72ad080703           /usr/lib/x86_64-linux-gnu/libstdc++.so.6+00386819
[  9] 0x00007f72ad080922           /usr/lib/x86_64-linux-gnu/libstdc++.so.6+00387362
[ 10] 0x00007f72ad080e0d           /usr/lib/x86_64-linux-gnu/libstdc++.so.6+00388621 _Znwm+00000125
[ 11] 0x00007f7178fe4311 /home/sdafarra/codyco-superbuild/build/install/mex/../lib/libYARP_dev.so.1+00877329
[ 12] 0x00007f7178feac24 /home/sdafarra/codyco-superbuild/build/install/mex/../lib/libYARP_dev.so.1+00904228
[ 13] 0x00007f7178fe26c4 /home/sdafarra/codyco-superbuild/build/install/mex/../lib/libYARP_dev.so.1+00870084
[ 14] 0x00007f717b8350a1 /home/sdafarra/codyco-superbuild/build/install/mex/../lib/libYARP_OS.so.1+01073313 _ZN4yarp2os4impl20PortReaderBufferBase4readERNS0_16ConnectionReaderE+00000337
[ 15] 0x00007f717b833e46 /home/sdafarra/codyco-superbuild/build/install/mex/../lib/libYARP_OS.so.1+01068614
[ 16] 0x00007f717b814afe /home/sdafarra/codyco-superbuild/build/install/mex/../lib/libYARP_OS.so.1+00940798 _ZN4yarp2os4impl8PortCore9readBlockERNS0_16ConnectionReaderEPvPNS0_12OutputStreamE+00000158
[ 17] 0x00007f717b822a9a /home/sdafarra/codyco-superbuild/build/install/mex/../lib/libYARP_OS.so.1+00998042
[ 18] 0x00007f717b892995 /home/sdafarra/codyco-superbuild/build/install/mex/../lib/libYARP_OS.so.1+01456533
[ 19] 0x00007f7179c50b16                           /usr/lib/libACE-6.0.3.so+00940822 _ZN21ACE_OS_Thread_Adapter6invokeEv+00000134
[ 20] 0x00007f72ac8f0184              /lib/x86_64-linux-gnu/libpthread.so.0+00033156
[ 21] 0x00007f72ac61d37d                    /lib/x86_64-linux-gnu/libc.so.6+01024893 clone+00000109
[ 22] 0x0000000000000000                                   <unknown-module>+00000000

If this problem is reproducible, please submit a Service Request via:
    http://www.mathworks.com/support/contact_us/
[... 96 lines cut ...]
yarp: Receiving input from /icubGazeboSim/right_leg/stateExt:o to /model/right_leg/stateExt:i using udp
yarp: Removing output from /model/torso/rpc:o to /icubGazeboSim/torso/rpc:i
yarp: Removing output from /model/torso/command:o to /icubGazeboSim/torso/command:i
yarp: Removing input from /icubGazeboSim/torso/state:o to /model/torso/state:i
yarp: Removing input from /icubGazeboSim/torso/stateExt:o to /model/torso/stateExt:i
yarp: Removing output from /model/left_arm/rpc:o to /icubGazeboSim/left_arm/rpc:i
yarp: Removing output from /model/left_arm/command:o to /icubGazeboSim/left_arm/command:i
yarp: Removing input from /icubGazeboSim/left_arm/state:o to /model/left_arm/state:i
yarp: Removing input from /icubGazeboSim/left_arm/stateExt:o to /model/left_arm/stateExt:i
yarp: Removing output from /model/right_arm/rpc:o to /icubGazeboSim/right_arm/rpc:i
yarp: Removing output from /model/right_arm/command:o to /icubGazeboSim/right_arm/command:i
yarp: Removing input from /icubGazeboSim/right_arm/state:o to /model/right_arm/state:i
yarp: Removing input from /icubGazeboSim/right_arm/stateExt:o to /model/right_arm/stateExt:i
yarp: Removing output from /model/left_leg/rpc:o to /icubGazeboSim/left_leg/rpc:i
yarp: Removing output from /model/left_leg/command:o to /icubGazeboSim/left_leg/command:i
yarp: Removing input from /icubGazeboSim/left_leg/state:o to /model/left_leg/state:i
yarp: Removing input from /icubGazeboSim/left_leg/stateExt:o to /model/left_leg/stateExt:i
yarp: Removing output from /model/right_leg/rpc:o to /icubGazeboSim/right_leg/rpc:i
yarp: Removing output from /model/right_leg/command:o to /icubGazeboSim/right_leg/command:i
yarp: Removing input from /icubGazeboSim/right_leg/state:o to /model/right_leg/state:i
```

I played a modified version of torqueBalancing, but I think that my modifications doesn't affect "initialization" phase. Anyway it is the first and only time it happened.

Thank you!

traversaro, 2016-08-22:
I strongly suspect this may be related to one of this problems : https://github.com/robotology/yarp/issues/880 , or many similar problems that have been fixed in `devel`.

gabrielenava, 2016-09-22:
I report that the same error occasionally happens on Ubuntu 16.04 with matlab 2016a.

gabrielenava, 2019-02-28:
To the best of my knowledge, this error never occurred with the latest versions of Simulink controllers in [whole-body-controllers](https://github.com/robotology/whole-body-controllers). Closing.

closed 2019-02-28 by gabrielenava