robotology-legacy/WBI-Toolbox-controllers #33: Simulink-Gazebo crash combos

gabrielenava, 2016-09-27:
Working with Gazebo simulator and Simulink controller I found some bugs related to either Gazebo, or Simulink, or other programs used for the simulation (wholeBodyDynamics3, yarpmotorgui, ...). I decided to list here all the "combos" that make Gazebo/Simulink/other stuff crash. 

- If I delete the model from gazebo while wholeBodyDynamics3 is still running, and I try to insert the model again, the new robot disappear from gazebo. If I stop the wholeBodyDynamics3, I can insert the new model without problems. There are no error or warnings related to this issue.
- Remove the robot (with yarpmotorgui running), then select a new one without inserting it in the Gazebo world. Then, if you close the yarpmotorgui, Gazebo crashes.
- If you remove the robot from Gazebo before stopping the simulation, Simulink crashes (this is more a warning than an issue)

Please update this issue if you find a new combo.

gabrielenava, 2017-10-06:
it seems to me these crash are not happening anymore. Closing

closed 2017-10-06 by gabrielenava

traversaro, 2017-10-06:
: D