robotology-legacy/WBI-Toolbox-controllers #36: Obtain geometry of the contact/support surface from the iCub skin

traversaro, 2016-12-22:
Until now, the geometry of the contact surface (typically the sole of the feet) has been hardcoded in the controllers using the `phys.footSize` and `gain.footSize` parameters [1], that express the support area as a rectangle in the `l_sole` or `r_sole` frames. This information is important 
because it imposes the set of physical contact force/torques that the robot can exert on the contact surface without slipping or tipping. 

For the CoDyCo Y4 Demo, we needed to balance the robot with the `l_upper_leg`/`r_upper_leg` links that are in contact with a side of the structural beam profiles, such as this one:
[image]

 As the beam profile is in contact only with part of the cover of the `*_upper_leg` link, we need to estimated online with part of the leg is actually in contact with the beam profile. This can be done with the iCub skin, but need some custom software. 

[1] See for example : https://github.com/robotology-playground/WBI-Toolbox-controllers/blob/e88439c24bcaecd3f2bfad4629cf359073a154ea/controllers/torqueBalancing/app/robots/iCubGenova02/gains.m#L122 .

iron76, 2016-12-27:
👍 When you write "with part of the leg is actually in contact" you mean "which area of the leg is actually in contact"?

gabrielenava, 2019-02-28:
After discussing with @Yeshasvitvs, we decided to close this issue as achieving a precise contact area size is not critical for the controller and the empirical size estimated by hand-measurements is enough for  now. Closing

closed 2019-02-28 by gabrielenava