robotology-legacy/WBI-Toolbox-controllers #42: Toolbox for exporting Simulink model in previous version automatically

gabrielenava, 2017-03-17:
Given the number of different Simulink versions we are using, this can be useful to ensure all users can access the most updated version of the code.

francesco-romano, 2017-03-17:
You can try if you are able to make [this](https://github.com/robotology/WB-Toolbox/blob/master/toolbox/export_library.m) work

See https://github.com/robotology/WB-Toolbox/issues/1

gabrielenava, 2017-04-06:
first version of this toolbox presented in branch `icub_standUp` and working fine. I will soon test it on master

closed 2017-09-26 by gabrielenava

traversaro, 2017-09-26:
@gabrielenava What is the status of this issue? The tool is working?

gabrielenava, 2017-09-26:
The function for exporting the model has been added to master since a while (https://github.com/robotology-playground/WBI-Toolbox-controllers/blob/master/controllers/torqueBalancing/exportMdl.m).  I tested it in iCubStandup walkman@rbc branches, and I remember I tested it before adding it to master, even if I cannot verify right now the effectiveness of the script.

francesco-romano, 2017-09-27:
Yes, the script works like a charm for model files. 
For some reasons I was never able to make it work for libraries