robotology-legacy/WBI-Toolbox-controllers #46: icubGazeboSim and iCubGenova02 models have frames oriented differenlty

gabrielenava, 2017-03-28:
There is a mismatch between the iCub model used for simulations and the one of iCubGenova02. The frames `l_upper_leg contact` and `r_upper_leg_contact` are oriented differently, as it is possible to see in the following images:

[image]
[image]

these frames are currently used only for `iCub stand up` simulation. Since `l_upper_leg_contact` at the initial time is taken as inertial reference frame, this mismatch causes the CoM references to have different signs in simulations and on the robot. The "correct" orientation is the one of icubSim, because it is the same of `l_sole` (while sitting) and therefore in this way all demos (Yoga, Coordinator, Standup) have the same references signs. Since I am for now the only user of iCub stand up demo on iCubGenova02, I will take care of regenerating the frames from CAD.

gabrielenava, 2017-04-06:
Corrected for the 4th year CoDycCo demo. Closing this issue

closed 2017-04-06 by gabrielenava

traversaro, 2017-04-06:
@gabrielenava Please note that the related issue https://github.com/robotology-playground/icub-model-generator/issues/42 is still open and should be address to make sure that all the CoDyCo partners can correctly run the CoDyCo Y4 Demo.

gabrielenava, 2017-04-06:
I'm aware of this issue, and I will start working on this more or less now (still 1 issue to fix in WBIToolbox-controllers before changing github repo)