robotology-legacy/WBI-Toolbox-controllers #51: WBToolbox: InverseKinematics box cannot retrieve string from robot part parameter

MichaelBombile, 2017-04-18:
Hi, we are trying to implement a reactive walking controller on iCub2 at EPFL LASA, and to do so we are trying to combine it with the matlab torqueBalancing module by exploiting the ComDes:i and the qDes:i input ports. However in our implementation we tried to add an inverse Kinematics box from the WBT_toolbox in order to compute the qDes and it keeps on giving the following error msg: 
**Error:Error reported by S-function 'WBToolbox' in 'torqueBalancingAndWalking/Inverse Kinematics1/S-Function': [mdlStart] Cannot retrieve string from robot part parameter**.

Here are the configuration parameters during the compilation:

[INFO][robotStatus::robotConfig] Configuring...
[INFO][robotStatus::robotConfig] wbiConfFileFullPath is: /home/michael/icub/codyco-superbuild/build/install/share/codyco/robots/icubGazeboSim/yarpWholeBodyInterface.ini
[INFO][robotStatus::robotConfig] List of properties for yarpWholeBodyInterface was interpreted correctly
[INFO][robotStatus::robotConfig] robot_fixed is false!
[DEBUG][robotStatus::robotConfig] After reading from config file, params are
[DEBUG][robotStatus::robotConfig] robot name: icubGazeboSim
[DEBUG][robotStatus::robotConfig] local name: matlabTorqueBalancing
[DEBUG][robotStatus::robotConfig] world reference frame: l_sole
[DEBUG][robotStatus::robotConfig] WBI ID List: ROBOT_TORQUE_CONTROL_JOINTS_WITHOUT_PRONOSUP

Please, can anyone tell us what is wrong and how we can fix it?

Thanks in advance.

DanielePucci, 2017-04-19:
Hi @MichaelBombile, @pi-q is going to provide with some support. 

@traversaro

pi-q, 2017-04-19:
Hi @MichaelBombile,

There has been some changes to WB-Toolbox committed by @francesco-romano a few hours ago to address this issue. You can update codyco-superbuild and see if you still get the error.

I personally tried using the inverseKinematics bloc after the update, but what I got as a result was Matlab crashing when starting the simulation. Maybe @francesco-romano could take a look?

DanielePucci, 2017-04-19:
Hi @pi-q, 
probably adding the MATLAB log that you get after the crash may help understand the problem.

MichaelBombile, 2017-04-19:
Hi @DanielePucci,
Thanks for your response.

I also had the same problem after the update. Matlab is now crashing. I restarted everything, rebuilt the toolbox but Matlab is still crashing.
I hope @francesco-romano can help

MichaelBombile, 2017-04-19:
Here is the crash report:

------------------------------------------------------------------------
       Segmentation violation detected at Wed Apr 19 14:50:05 2017
------------------------------------------------------------------------

Configuration:
  Crash Decoding      : Disabled - No sandbox or build area path
  Crash Mode          : continue (default)
  Current Graphics Driver: Unknown software 
  Current Visual      : 0x21 (class 4, depth 24)
  Default Encoding    : UTF-8
  Deployed            : false
  GNU C Library       : 2.23 stable
  Host Name           : lasapcMichael
  MATLAB Architecture : glnxa64
  MATLAB Entitlement ID: 815978
  MATLAB Root         : /usr/local/MATLAB/R2016b
  MATLAB Version      : 9.1.0.441655 (R2016b)
  OpenGL              : software
  Operating System    : Linux 4.4.0-70-generic #91-Ubuntu SMP Wed Mar 22 12:47:43 UTC 2017 x86_64
  Processor ID        : x86 Family 6 Model 58 Stepping 9, GenuineIntel
  Virtual Machine     : Java 1.7.0_60-b19 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode
  Window System       : The X.Org Foundation (11804000), display :0

Fault Count: 4

Abnormal termination:
Segmentation violation

Register State (from fault):
  RAX = 00007fba794d0420  RBX = 0000000000000001
  RCX = 9c64bfb448342018  RDX = 00007fbc0ea53da0
  RSP = 00007fbc0ea536c0  RBP = 138c87ff3fffffff
  RSI = 00007fba4a711490  RDI = 0000000040000000

   R8 = 2ad3400000000018   R9 = 0000000000000000
  R10 = 0000000800000008  R11 = 639bc00000000008
  R12 = 0000001800000018  R13 = 0000000100000001
  R14 = 00007fba48342020  R15 = 3ffffffe3fffffff

  RIP = 00007fba794d09aa  EFL = 0000000000010216

   CS = 0033   FS = 0000   GS = 0000

Stack Trace (from fault):
[  0] 0x00007fba794d09aa        /usr/local/MATLAB/R2016b/bin/glnxa64/mkl.so+23271850

Abnormal termination:
Segmentation violation

Register State (from fault):
  RAX = 00007fba794d0420  RBX = 0000000000000001
  RCX = 9c64bfb848342018  RDX = 00007fbc0ea53da0
  RSP = 00007fbbe32667c0  RBP = 138c87ffbfffffff
  RSI = 00007fbe4a711490  RDI = 0000000040000000

   R8 = 2ad3400000000018   R9 = 8000000080000000
  R10 = 0000000800000008  R11 = 639bc00000000008
  R12 = 0000001800000018  R13 = 0000000100000001
  R14 = 00007fba48342020  R15 = bffffffebfffffff

  RIP = 00007fba794d09a6  EFL = 0000000000010216

   CS = 0033   FS = 0000   GS = 0000

Stack Trace (from fault):
[  0] 0x00007fba794d09a6        /usr/local/MATLAB/R2016b/bin/glnxa64/mkl.so+23271846

Abnormal termination:
Segmentation violation

Register State (from fault):
  RAX = 00007fba794d0420  RBX = 0000000000000001
  RCX = 00007fba48342020  RDX = 00007fbc0ea53da0
  RSP = 00007fbbe0982ac0  RBP = 0000000000000000
  RSI = 00007fc04a711490  RDI = 0000000040000001

   R8 = 2ad3400000000018   R9 = c0000000c0000000
  R10 = 0000000800000008  R11 = 639bc00000000008
  R12 = 0000001800000018  R13 = 0000000100000001
  R14 = 00007fba48342020  R15 = 0000000000000000

  RIP = 00007fba794d09a6  EFL = 0000000000010212

   CS = 
If this problem is reproducible, please submit a Service Request via:
    http://www.mathworks.com/support/contact_us/

A technical support engineer might contact you with further information.

Thank you for your help.** This crash report has been saved to disk as /home/michael/matlab_crash_dump.7840-3 **

	-------------------------------------------------------------
	This segmentation violation occurred while executing the 
	S-function 'WBToolbox' in block 't'.
	A common cause of this segmentation violation is an incorrect
	input port direct feedthrough setting. Each input port of the
	S-function that is read (accessed) in mdlOutputs and/or
	mdlGetTimeOfNextVarHit must specify that it needs its input
	signal in these routines by setting direct feedthrough for
	these input ports.
	Another cause can be incorrect memory accesses which occur
	when your code accesses beyond the end of an array. For example
	if you access input port 5 and in mdlInitializeSizes specify
	that you only have 4 input ports.
	To debug your C-MEX S-function, you can enable diagnostics
	by compiling the S-function source with the -g flag, e.g.,
	  mex -g sfunction_name.c
	-------------------------------------------------------------

DanielePucci, 2017-04-19:
Mmmmmmm the stack trace does not seem to help. @pi-q, can you do an in-depth analysis and understand the problem?

francesco-romano, 2017-04-19:
Sorry @pi-q and @MichaelBombile 
I am not able to replicate the crash.

I will need further information to debug this issue.

pi-q, 2017-04-20:
Could it be a problem related to Ubuntu vs Apple or the version of Matlab used? Using Matlab 2016b on Ubuntu, I get the same crash as @MichaelBombile when trying to run a simulation containing nothing more than the InverseKinematics block.

Simulink's diagnostic viewer adds this short line:
```
An error occurred while running the simulation and the simulation was terminated
Caused by:
FatalException
```

francesco-romano, 2017-04-20:
Yes. It could be.
Unfortunately the error message does not help.

Can you try to compile in debug to see if a more verbose message is printed?

pi-q, 2017-04-20:
Here would be the verbose output related to InverseKinematics when I compile in debug:

```
[ 43%] Building CXX object toolbox/CMakeFiles/WBToolbox.dir/src/InverseKinematics.cpp.o
cd /home/marie/Programs/codyco-superbuild/main/WBToolbox/build/toolbox && /usr/bin/c++   -DDLL_EXPORT_SYM="__attribute__ ((visibility (\"default\")))" -DMATLAB_MEX_FILE -DWBT_USES_ICUB -DWBT_USES_IPOPT -isystem /home/marie/Programs/icub-main/src/libraries/iKin/include -isystem /home/marie/Programs/icub-main/src/libraries/ctrlLib/include -isystem /home/marie/Programs/codyco-superbuild/main/WBToolbox/toolbox/autogenerated/include -I/home/marie/Programs/codyco-superbuild/main/WBToolbox/toolbox/include -I/home/marie/Programs/codyco-superbuild/main/WBToolbox/toolbox/include/base -isystem /usr/local/MATLAB/R2016b/extern/include -isystem /usr/local/MATLAB/R2016b/simulink/include -isystem /home/marie/Programs/yarp/build/generated_include -isystem /home/marie/Programs/yarp/src/libYARP_OS/include -isystem /home/marie/Programs/yarp/src/libYARP_sig/include -isystem /home/marie/Programs/yarp/src/libYARP_eigen/include -isystem /home/marie/Programs/yarp/src/libYARP_gsl/include -isystem /home/marie/Programs/yarp/src/libYARP_math/include -isystem /home/marie/Programs/yarp/src/libYARP_dev/include -isystem /home/marie/Programs/yarp/src/libYARP_name/include -isystem /home/marie/Programs/yarp/src/libYARP_manager/include -isystem /home/marie/Programs/yarp/src/libYARP_logger/include -isystem /home/marie/Programs/codyco-superbuild/build/install/include -isystem /usr/include/eigen3 -isystem /home/marie/Programs/icub-main/src/libraries/skinDynLib/include -isystem /home/marie/Programs/codyco-superbuild/build/install -isystem /home/marie/Programs/icub-main/src/libraries/iDyn/include  -g -fPIC -fvisibility=hidden -fvisibility-inlines-hidden   -pthread -std=gnu++11 -o CMakeFiles/WBToolbox.dir/src/InverseKinematics.cpp.o -c /home/marie/Programs/codyco-superbuild/main/WBToolbox/toolbox/src/InverseKinematics.cpp
```

Do you find any clue in there?

francesco-romano, 2017-04-20:
thanks @pi-q.
What I meant is if with the library compiled in debug the crash is more verbose (even if I doubt).

Anyway, can you also paste the linking phase (not only the compile phase) of the library

MichaelBombile, 2017-04-21:
Hi @pi-q, 
Is there any progress in the solving this the inverse Kinematics issue? You have just closed this issue and could you please let me know if you managed to fix the problem.

pi-q, 2017-04-21:
Sorry I didn't mean to close the issue, that was an accident. Still no breakthrough has been reached.

@francesco-romano, I didn't see any difference in the crash log after compiling in debug.

I am not sure I get what you mean about the linking phase, the only thing I get related to linking during compilation is this line:
`[100%] Linking CXX shared library WBToolbox.mexa64`

MichaelBombile, 2017-04-21:
Ok, thanks

francesco-romano, 2017-04-21:
then I am afraid i am not able to solve this with this few amount of information

pi-q, 2017-04-21:
As I see it, Simulink goes through the code to initialize the block, but is unable to go any further. More than that, I don't know what is going on. Maybe I am not the best person to figure it out?

MichaelBombile, 2017-04-21:
I think the problem could come from the way the iKinIpOptMin solver is used in the implementation of the toolbox. As I was going through the code, I tried to comment the lines 309 to 313 (where the solver is called in the output function)  Simulink ran normally and gave of course the input joints angles as output. 
But when I uncommented them, Matlab crashed again.
Could this give you a clue on the problem?

I should mentioned that when the solver iKinIpOptMin is tested in other codes, it  works perfectly.

MichaelBombile, 2017-04-25:
Hi @pi-q, did my last comment help in one way or another to figure out what the problem could be?
Is there any progress in addressing this issue?

pi-q, 2017-04-28:
Hi @MichaelBombile,  I haven't had the chance to progress on this issue. Hopefully next week I will manage to take a look.

MichaelBombile, 2017-05-09:
Hi @pi-q, did you have time, last week, to look into the Inverse Kinematics toolbox issue?

gabrielenava, 2019-02-28:
The issue is related to a code which is stored in a different repository: [wb-toolbox](https://github.com/robotology/wb-toolbox). Also, this repository is no  more maintained, therefore if the problem still occurs please re-open this issue in [wb-toolbox](https://github.com/robotology/wb-toolbox). Closing.

closed 2019-02-28 by gabrielenava