robotology-legacy/WBI-Toolbox-controllers #52: All balancing controllers fail to initialize

gabrielenava, 2017-04-26:
I updated yarp, icub-main and codyco-superbuild to the latest versions. Then, I run (not just compile) `torqueBalancing.mdl`. I got the following error:

```
[ERROR]yarpWholeBodyStates::init : estimator thread initialization failed 
[ERROR][mdlStart] Failed to initialize WBI 
```

this occurs both in `master` and `devel` branch, and for all controllers (e.g. iCub, Walkman, ...) and also for this simple test:

[image]

this is what is going on in Matlab terminal:

```
[ERROR]yarpWholeBodySensors::readPwms failed for timeout
yarp: A thread failed to start with error code: 0
[ERR] Error while initializing yarpWholeBodyStates interface.
```

no other error messages are given. Tested on Ubuntu 16.04 LTS.

traversaro, 2017-04-26:
Related line of code : https://github.com/robotology/yarp-wholebodyinterface/blob/c9b9ba509ba56f7ca911e35f2e22e79285a8ae72/src/yarpWholeBodySensors.cpp#L725 . 
Can you check if this affects also a simple C++ example using the `yarpWholeBodyInterface`, for example: https://github.com/robotology/yarp-wholebodyinterface/blob/master/examples/yarpWholeBodyInterfaceExample/main.cpp .

nunoguedelha, 2017-05-03:
As @gabrielenava  observed and verified by reverting the iopl (interface to read motor PWM) type to `std::vector<IOpenLoopControl*>`, it seemed there was something wrong going on with the casting in line https://github.com/robotology/yarp-wholebodyinterface/blob/c9b9ba509ba56f7ca911e35f2e22e79285a8ae72/src/yarpWholeBodySensors.cpp#L716.

```
#ifndef YARPWBI_YARP_HAS_LEGACY_IOPENLOOP
        while( !(update=((IPWMControl*)iopl[*ctrlBoard])->getDutyCycles(pwmTemp)) && wait)
#else
        while( !(update=((IOpenLoopControl*)iopl[*ctrlBoard])->getOutputs(pwmTemp)) && wait)
#endif
```

Analysing the code with the debugger while running the example @traversaro  mentioned confirms that indeed `getOutputs(pwmTemp)` is never called. Actually the cast to `IOpenLoopControl*` is done in some way but doesn't allow the polymorphism to work properly: we're not pointing to `getOutputs(pwmTemp)` of the class `yarp::dev::RemoteControlBoard` object (device driver implementing the interface). The issue is actually happening earlier, in `yarpWholeBodySensors::openPwm(const int bp)` and understanding what happens there would explain the problem...

1 - in this function, the iopl pointer is passed to the device driver that implements the interface:
`if(!dd[bp]->view(iopl[bp])) ...`
The `view` function is using a dynamic cast, requiring RTTI (RunTime Type Information), to cast originally the pointer to the device driver (`yarp::dev::RemoteControlBoard`) into a `IOpenLoopControl*` pointer.

2 - as `yarp::dev::RemoteControlBoard` implements a `IOpenLoopControl`, the polymorphism should do the work (no need for casting) and should allow a pointer `IOpenLoopControl*` to access the implemented method `getOutputs(...)`.

The issue is that that dynamic cast doesn't work properly on a (void*) pointer, and by the time get to the function `yarpWholeBodySensors::readPwms(...)` the polymorphism seems to be "broken".

### Proposed solution still keeping a `void* iopl`:
If we use an explicit type `(IPWMControl*)` or `(IOpenLoopControl*)` by the time the dynamic cast is done, and do a normal cast later into `(IPWMControl*)` or `(IOpenLoopControl*)`, everything works fine. I pushed a fix

traversaro, 2017-05-03:
Thanks @nunoguedelha ! 
I open a revert PR and cherry-picked your fix in https://github.com/robotology/yarp-wholebodyinterface/pull/77 .

closed 2017-05-04 by nunoguedelha