robotology-legacy/WBI-Toolbox-controllers #57: Update documentation of the repo

gabrielenava, 2017-09-07:
It seems to me the README.md files in teach controller folder are outdated. An example:

in this [README](https://github.com/robotology-playground/WBI-Toolbox-controllers/blob/master/controllers/torqueBalancing/README.md), it is stated that one must run `wholeBodyDynamicsTree` in simulation, which has been substituted by `wholeBodyDynamics`. Also `twoFeetStandingIdleAndCalib` is not used anymore in simulation, and it is substituted by sending `resetOffset all` command to wholeBodyDynamics.

gabrielenava, 2017-10-03:
As noted by [@yeshasvitvs](https://github.com/Yeshasvitvs), there are also comments inside matlab functions that are not updated (e.g. wrong files names, and so on). Therefore we're going to fix these typos and proceed with the cleanup.

Yeshasvitvs, 2017-10-03:
@gabrielenava I cleaned up the typos and opened a [PR](https://github.com/robotology-playground/WBI-Toolbox-controllers/pull/60). Please review it.

gabrielenava, 2017-10-06:
done with https://github.com/robotology-playground/WBI-Toolbox-controllers/commit/96fd372de2257bf603ea2a4f7dfef6581b631d85

closed 2017-10-06 by gabrielenava