robotology-legacy/WBI-Toolbox-controllers #62: Should we remove old configuration files?

gabrielenava, 2017-10-05:
Inside the [app/robots](https://github.com/robotology-playground/WBI-Toolbox-controllers/tree/master/controllers/torqueBalancing/app/robots) folder of `torqueBalancing` and `wholeBodyImpedance` controllers there are configuration files for different robots. I think some of them are quite old and probably they are not even usable anymore. In particular:

#### torqueBalanging

- iCubDarmstadt01
- iCubGenova01
- iCubGenova03
- iCubGenova05
- iCubParis01
- iCubParis01

#### wholeBodyImpedance

- iCubDarmstadt01

should we remove them? also considering that even if deleted we can recover them if necessary looking at github history.

cc @DanielePucci

DanielePucci, 2017-10-05:
Why can't we keep them?

gabrielenava, 2017-10-06:
I opened them and they probably cannot work anymore because of missing variables, therefore I think that if we're going to work on one of these robots best solution will be to copy-paste configuration files from another one (probably iCubGenova04). But I agree that removing those files may be considered "premature optimization" [cit.]

traversaro, 2017-10-06:
I agree in removing those files, also to clarify which platform we are supporting and which platform we are not supporting.

gabrielenava, 2017-10-06:
ok then! I'm going to remove these files.

gabrielenava, 2017-10-06:
done with https://github.com/robotology-playground/WBI-Toolbox-controllers/commit/96fd372de2257bf603ea2a4f7dfef6581b631d85

closed 2017-10-06 by gabrielenava